tardate/LittleArduinoProjects #2: Re #156 MSO5074FG_unboxing

NikolaiRuhe, 2017-12-04:
Thanks a lot on your review of the Hantek MSO5074FG.

After a couple of months, are there any additional insights on the scope? Would you buy it again? Would you recommend a different device?

Thanks a lot for your time!
Nikolai

tardate, 2017-12-05:
Hi @NikolaiRuhe, I've had the scope for maybe a year now. I've learned a few quirks and limitations, but overall I've been very happy with it. If price and feature differentials haven't changed in the meantime, I'd  buy it again.

The main thing that attracted me: if you are only going to spend << $1000, and this will be the only scope on the bench for some time, the features (4-channel, and the logic analyser) are not going to be limiting unless you are pushing into more specialised fields. In particular, I am really glad I did not compromise on 2-channel in order to fit a "brand name" scope in the budget.

The main thing that bugs me: screen captures from the scope include the screen capture menu, and I've never found a way to avoid this! I've also had it crash and reboot on me a few times - seems to occur when I am very quickly reconfiguring many channel and trigger settings - but it is fine after the reboot and no big deal, and only happened maybe 4 times in a year.

So, yes bottom line: very happy I went with this scope at the time. 

PS: my opinions totally anecdotal based on a sample size of 1 of course

closed 2017-12-05 by tardate

NikolaiRuhe, 2017-12-05:
Many thanks for your detailed answer!

I've been torn between the Hantek and some Rigol device. It's difficult to find proper advice in this low budget sector but your article and now your answer gave me the final push.

I'm planning to use it mainly for hobby projects and maybe some work related µC firmware stuff. From your blog it looks like you're using the scope in a very similar way so I value your opinion.

Thanks again and best greetings from Berlin!
Nikolai

tardate, 2017-12-05:
Cheers @NikolaiRuhe, hth although I'm always wary of providing advice - this is just a hobby for me too! 

I remember when I made the purchase the debate in my head was basically: "I really want the features of the Hantek, but what if it is a dud? Maybe I should compromise on features (2 instead of 4 channels) and play it safe with a Rigol". Well, it turned out just fine for me - Hantek may not be perfect but they are definitely not duds.

And one thing is always true: any scope is better than no scope;-)