tardate/LittleArduinoProjects #7: The D2-1 kit is more simple to mount

zoomx, 2018-10-19:
Hello,
I have just read the D2-5 kit pages. I mounted this kit but I had also the D2-1 kit, that has about the same price and is very similar, the only thing different are the motors.
It is more simple to mount because you don't have to align motors, gear wheels and axles.

tardate, 2018-10-19:
thanks for the info @zoomx. I've added a mention to https://github.com/tardate/LittleArduinoProjects/tree/master/Kinetics/LineFollowerKit

tardate, 2018-10-19:
have you built the D2-1 kit @zoomx? My first thought with this kit is that it would be a great project to get someone engaged beyond just flashing LEDs. Big payoff (in terms of fun) for just a little bit of skill to build;-)

closed 2018-10-19 by tardate

zoomx, 2018-10-19:
I build both.
I speak briefly about D2-1 here
[https://www.stm32duino.com/viewtopic.php?f=50&t=2198](url)
(there are errors about kit identification!)
I made some brain surgery and used an STM32F103 instead of the LM393, using the same battery, so you have a very cheap robotic platform. I tried also ESP8266 but the power isn't enough even if a use a step-up board.
It is very simple to build, more than the D2-5, it is very cheap and I believe that you can convert to a little robotic platform very easy.

tardate, 2018-10-19:
thanks, I've linked your post in the doc too;-)