tardate/LittleArduinoProjects #11: MilliVoltmeterDIY - accuracy

alex1sh, 2019-01-12:
Hello,

I have assembled the MilliVoltmeter, but sadly accuracy is a bit off. When I calibrate it against 2.5V ref, I then measure 5V and 10V voltage standards. On 5V (gives me 5.000001v on the 6 1/2 DMM) I get 4.97V and on 10V I get 9.91V. Any idea why? I have used resistors with 1% tolerance. 

On the Scullcom MilliVoltmeter, I do not have this issue. It is very precise and accurate. 
What's accuracy on your meter please?

Thanks
Alex

tardate, 2019-01-13:
hi @alex1sh hmm, you're right. I've updated my notes with some measurements and more. I'm seeing skewed results from the LTC2400 far beyond the expected error rates from the datasheet. I don't yet see why that should be the case .. do you have any suggestions?

alex1sh, 2019-01-14:
Hi Paul. I think the problem is in the input section of your schematic. You are using 1Mohm and 100Kohm resistors for a voltage divider. Say, I applied 10.000 volts to VIN. According to the voltage divider calculation, I am not getting VINx10 (1 volts), but 0.909 volts. I will try to change 1Mohm resistor for 900Kohm one. This is give me exactly 1/10 factor. I have used both resistors with tolerance of 1%, but really they have to be 0.05% tolerance or better and low ppm as this may give you different readings for the same source. Thanks Alex

tardate, 2019-01-14:
The values of the voltage divider should not be critical @alex1sh  .. as long as it is a stable (stiff) fraction R1/(R1+R2), that constant is accommodated in the calibration factor. i.e. when calibrated with the reference voltage (2.5V), the ADC value represents R1/(R1+R2) * 2.5

Boundary case is 0V i.e. VIN=0V, which I'd expect the ADC to report as 0, but I'm seeing ~0x4666 (27mV). I think I'll thrown an LT1019 and LTC2400 on a breadboard again and dive in a bit deeper; right now I'm distrusting the results I'm seeing from the LTC2400!

tardate, 2019-01-16:
btw @alex1sh I've been doing some investigations (which I'll write-up separately soon). Bottom line - seems I was a bit quick to just slap a voltage-divider directly on the LTC2400 input (it doesn't just act as a high-impedance passive load). The 1MΩ/100kΩ is "not bad", but the LTC2400 is meant to be capable of better, however it looks like I need a more sophisticated front-end to achieve that.

alex1sh, 2019-01-17:
@tardate 

Have a look at the attachment. I do not see a problem attaching the voltage divider directly onto the LTC2400 input.

[image]

alex1sh, 2019-01-18:
@zoomx There is a voltage divider (100k / 900k - I have replaced 1m resistor with 900k) before LTC2400. This divides voltage by 10. Therefore, 10V becomes 1V etc.

zoomx, 2019-01-18:
@alex1sh 
I forgot that this project uses an LTC2400 and not the Atmega328 ADC!

tardate, 2019-01-18:
@alex1sh thanks, great video. But I really hope its possible to avoid multi-step calibration!

The demo board schematic certainly seems to imply no problem with a big cap on the input and potentially a voltage divider. But I've been reading p22 "Driving the Input and Reference" very carefully, and it seems to tell another story.

That's what I'm seeing experimentally too. I'm going to push my notes/wip tests when I get a chance so you can see what I have/haven't tested in more detail. 

Re:

> 1) Vref voltage has to be measured with 6.5 DMM and inserted into the sketch
yes, that could be key. Especially with a relatively low VREF, errors up at 10V etc are magnified.
I've got a 4.096 ref (like scullcom used) but haven't tried it yet to see if it is material

> 2) The voltage divider has to be more accurate
with the current calibration approach, actually, I think no. The ratio should not have any impact on the result,  as whatever it happens to be, the calibration step captures it. The only issue is that with that one data point, it assume straight-line through a 0 origin; any skew and offset errors of course undermine that assumption. 

> 3) non-linearity of LTC2400 has to be dealt with
yes! I'm slowly coming to the conclusion that a buffer opamp may be an unavoidable necessity. I really preferred not to do that, because I also wanted true 0 reading without needing a dual-rail power supply (I guess the same reason the precision voltage ref in the vid starts at 50mV;-) 

I'l update here when I've pushed my notes and you can take a look (might be later this weekend as I have a few other things to take care of).

@zoomx for VREF=2.5V, I'm expecting the extended input range should – 0.3V to 1.125 • VREF (per Note 14, p4), but as @alex1sh says, with the voltage divider all of my/our testing has been well within that range. hmm, I don't see anything about `Vref-(Vref/1024)`, can you point it out?

alex1sh, 2019-01-20:
@ tardate I think you are correct - using AD820 opam or similar is unavoidable. 
Something like this. 
[image]

Unfortunately, I do not have a spare LTC2400 to test on the breadboard. I have a similar millivoltmeter designed by Scullcom with the buffer opamp and it words very-very precisely. FYG, the voltage divider is 0.1% tolerance and low tempco.

I think a small daughter board is required on the input here. This is not a bad thing as this analogue board will be completely isolated from the digital one.

alex1sh, 2019-01-20:
This is an idea for an analogue daughter board:

[image]

Works really precisely.

tardate, 2019-01-27:
@alex1sh I finally got around to pushing some test results and notes - see https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/ADC/LTC2400. Current situation: still failing to get decent results even with the most basic configuration. My next step is to try a better op-amp for the buffer. If you can make sense of my notes, I'd appreciate any comments you might have.

Cheers,
Paul

alex1sh, 2019-01-28:
@tardate Excellent notes I must admit. A week ago I have re-read LTC2400 datasheet and soldered 10uF cap to the voltage divider. To my surprise reading especially on the higher end, were completely off. So I cannot see how soldering this cap can help accuracy. Your datasheet findings just confirm it. 
In fact I am pretty much in line with your on the datasheet readings re Absolute Maximum Ratings.

I think the only way to achieve a good result is to use a buffered VIN for LTC2400. LMV324 is a general purpose OpAmp. I do not think this would be an appropriate part for this project. What you really need is a good zero-drift rail-to-rail OpAmp. Good example could be OPA388.  http://www.ti.com/lit/ds/symlink/opa388.pdf

I managed to find a few on Digi-Key:

https://www.digikey.co.uk/products/en/integrated-circuits-ics/linear-amplifiers-instrumentation-op-amps-buffer-amps/687?FV=284008a%2Cmu0.2%C2%B5V%7C975%2Cmu0.3%C2%B5V%7C975%2Cmu0.4%C2%B5V%7C975%2Cmu0.5%C2%B5V%7C975%2Cmu0.6%C2%B5V%7C975%2Cmu0.7%C2%B5V%7C975%2Cmu0.8%C2%B5V%7C975%2Cmu1%C2%B5V%7C975%2Cii1%7C2094%2Cffe002af&quantity=0&ColumnSort=975&page=1&stock=1&pageSize=500

The one chosen by Scullcom is a good one - AD8628ARZ. From its datasheet:
The AD8628/AD8629/AD8630 are single-supply, ultrahigh precision rail-to-rail input and output operational amplifiers. The typical offset voltage of less than 1 µV allows these amplifiers to be easily configured for high gains without risk of excessive output voltage errors. The extremely small temperature drift  of 2 nV/°C ensures a minimum offset voltage error over their entire temperature range of −40°C to +125°C, making **these amplifiers ideal for a variety of sensitive measurement applications in harsh operating environments.**

alex1sh, 2019-01-28:
I am going to order OPA388 now:

The OPAx388 (OPA388, OPA2388, and OPA4388) series of precision operational amplifiers are ultra-low noise, fast-settling, zero-drift, zero-crossover devices that provide rail-to-rail input and output operation. These features and excellent ac performance, combined with only 0.25 µV of offset and 0.005 µV/°C of drift over temperature, makes the OPAx388 ideal for driving high-precision, analog-to-digital converters (ADCs) or buffering the output of high-resolution, digital-to-analog converters (DACs). This design results in superior performance when driving analogto-digital converters (ADCs) without 
degradation of linearity

tardate, 2019-01-29:
@alex1sh yup, I'm hoping to see some big improvement with a precision op-amp. I've got some AD8628 on order, so we should be able to compare notes;-)

alex1sh, 2019-02-07:
Hi Paul,

I have tried using OPA388.
Sadly, I cannot see a much better performance - on 10.000000V reference I get 9.956889. I’m using precise resistors (0.1%) on the voltage divider 

Cheers
Alex

tardate, 2019-02-08:
@alex1sh I just received some precision op-amps incl AD8628 and OPA2388 and should get a chance to do some test over the weekend. I hope I can get some better results!

tardate, 2019-02-15:
fyi @alex1sh , added some more results for a test with a bunch of OPA2388 buffers, and starting to get some more decent results

* [OPA2388-buffered VREF](https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/ADC/LTC2400#scenario-4-opa2388-buffered-vref-signals) - tests the ability to read back the ref voltage, no voltage divider
* [OPA2388-buffered Everything](https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/ADC/LTC2400#scenario-5-opa2388-buffered-everything) - adds voltage divider

adding lots of bypass caps and using a stable dedicated power supply, I'm finally getting readings that are stable within 0.06%, and accurate down to the 4th digit (sub mA). 

There are more tests I want to do, and I haven't tested accuracy over the full voltage range yet. But I'm finally a little encouraged. I think the next I'll try is compare precision and stability with a 4.096V reference (as used in the scullcom design - I finally have an ADR4540 to test with)

alex1sh, 2019-02-17:
@tardate 

Hi Paul,

Interesting observations.
As we are trying to extract everything down to 6th digit, I think there may be a problem using a breadboard. The breadboard is just one big capacitor and I saw voltage varying on my bench DMM as I was changing configuration on the breadboard (same schematic). Did you try to solder parts together on a simple PCB? This should improve results. Re Vref - my latest research into this showing that a decent LT399 (10V, but can do 5V with INA105) or LTC6655-4.096 would be required for a low drift. Ground is extremely important on the analogue side.
I have not used bypass caps in my last tests and yes you are correct after a little tinkering I can confirm it improves the final result, i.e. gives more "clean" power supply.

I have got a new bench power supply Elektro-Automatik. It does have a very low ripple - I'll run tests once again when it arrives if I have time.

tardate, 2025-01-25:
It's been a minute... 

While we were still hunting better accuracy and stability, I'm going to close this issue out for now.

To be continued if/when this project gets a fresh look!

closed 2025-01-25 by tardate