tardate/LittleArduinoProjects #16: AudioEffectsChips/PX835 - Song names

whc2001, 2019-12-22:
Judging by the short sample video, I found that this chip has an identical one labeled "YSJ-M16" manufactured by YuSiJie Electronics. However I am not able to find any retailer for either model except on Chinese shopping site Taobao.

| No | Title |
|----|-------|
| 1  | Fur Elise |
| 2  | 虹彩妹妹 |
| 3  | [Ding Dong * 2] |
| 4  | Rhythm of the Rain |
| 5  | Jingle Bells |
| 6  | Romeo and Juliet |
| 7  | Love Story |
| 8  | 太湖船 |
| 9  | [Westminster Chime] |
| 10 | Oh My Darling Clementine |
| 11 | We Wish You a Merry Christmas |
| 12 | Oh Susanna |
| 13 | Wiegenlied |
| 14 | Silent Night |
| 15 | Turkish March |
| 16 | [Cuckoo * 3] |

[image]

tardate, 2019-12-22:
thanks again for the song titles and sourcing tip @whc2001; I found 宇思捷电子 listing on taobao and the **81** music chips they have listed(!!)

closed 2019-12-22 by tardate

whc2001, 2019-12-22:
> thanks again for the song titles and sourcing tip @whc2001; I found 宇思捷电子 listing on taobao and the **81** music chips they have listed(!!)

Yes, they focus on manufacturing these low-cost single-function chips.