tardate/LittleArduinoProjects #18: MT3608 problem

behnam68, 2020-10-08:
Hello dear friends
My designing pcb for MT3608 not working at all and there's no output boost voltage ( input and output are the same)
i attached my pcb to check it out.i dont find any difference between my pcb and other working modules.i also used the exact component of a working commercial module in my pcb and not working again.
Thanks for helping and excuse me for my bad english.
[image]
[image]

jjohnson873, 2020-10-08:
You may want to post a picture of your schematic, including the components values.   Make sure you buy your MT3608 from a trusted source.   I've seen other MT3608 commercial boards, that had counterfeit parts and the part behaved the same way as your board.  Did you try changing out the MT3608 for another part?  The other thing to check is the correct installation of the diode.  The diode should have the anode connected to pin#1 of MT3608 and the cathode to the board output.

behnam68, 2020-10-08:
@jjohnson873 thank you
This is my schematic:
[image]
I have some cheap Chinese ICs but i'm sure MT3608 that i bought is OK because when i replace the on-board  working module's IC with my IC, it works fine, but the same IC doesn't work on my designed board. for diode as you can see in the pictures i've posed the polarity and connection is according to schematic and your description. i measured the caps value on the two different commercial modules and one of them has two 4.7uf and the other has two 10uf caps so i've tried 10uf,4.7uf and 1uf 50v instead of 22uf ceramic caps and not working.

tardate, 2020-10-08:
hi @behnam68, I don't see anything obviously wrong with the circuit, PCB or the build. The fact that you are seeing Vout ~= Vin would tend to indicate there's no switching going on (could be verified with a scope on SW pin if you have one available). 

Assuming the MT3608 chip is ok, connections are good, especially that EN is a good connection to pull high, then this may just be the R2 configuration providing a misleading test result:

In potentiometer mode (as in your schematic), I don't think R2 is actually going to be effective over its full range:

- i.e.if IN+ is 5v, and given that FB is controlled to approach 0.6V
- then for lowest boost out, FB would be at the 0.6V/5V*102.2kΩ = 12kΩ mark
- and max boost out, FB would be at the 0.6V/28V*102.2kΩ = 2kΩ mark

So for most of its travel, R2 is going to mess with the FB level, and I guess the MT3608 will just cut out?

I've see this config in many schematics, but on the commercial boards I have and other schematics on the net, they use a "rheostat" config (ie the wiper of the pot tied to one end). I've also used this successfully for the test boards I've made e.g. https://leap.tardate.com/electronics101/switchmodepowersupplies/mt3608/variableboost/ 

In rheostat mode and IN=5V
- then min boost out, the R2 would be at (5V-0.6V)/0.6V*2.2kΩ = 16kΩ mark
- and max boost out, FB would be at (28V-0.6V)/0.6V*2.2kΩ = 100kΩ mark
So still a slight dead end, but much more of the travel should cause boost.
 
At least, that seems like a possibility, if I got my math right!
Good news is that you can try this simply by bridging the lower two pins of the R2 trimmer.

behnam68, 2020-10-08:
@tardate thank you so much for great info

> I've see this config in many schematics, but on the commercial boards I have and other schematics on the net, they use a "rheostat" config (ie the wiper of the pot tied to one end)

after remove two commercial module that i have , find that tips that you mention above in one of the modules. 
you can see the picture of them :

Module #1:
[image]
[image]

Module #2:
[image]
[image]

as you can see my design is similar to module#1 and what you mention is module#2 ... and both of modules works fine.

> Good news is that you can try this simply by bridging the lower two pins of the R2 trimmer.
exactly and i did it before but unfortunately nothing change in the output of my pcb.

I've test some different pcb desing (with the same schematic) such as one side , double sided metalized and simple , with ground plane or not .... and result of all are the same. i believe that something is wrong but i couldn't find it yet...design or parts?!

jjohnson873, 2020-10-08:
I've enlarged your photos and most all of the solder joints look ok.  I'd recommend you use flux next time since it doesn't appear you applied flux to the pc board before you soldered on the parts.   You'll use less solder and the connections will be better.   The Schottky diode appears correct on your board and you used an SS34 which is rated correctly.   The only two things I can think of is that the MT3608 got damaged or is faulty, or there may be a solder short under your inductor.   Also the inductor needs to be between 4.7uH to 22uH with low loss at 1.2MHz and a low DCR.   Agree with tardate, short the pot such that the wiper is shorted to the pin connected to your R1.  In this manner, you're not changing the lower resistance (R1) presented to FB as you turn the pot.  If you just had R2 fixed at 100K-ohm and R1 at 2.2K-ohm you should get a 28vdc output.  Did you ever say what input voltage you're using?

tardate, 2020-10-09:
> as you can see my design is similar to module#1 and what you mention is module#2

Thanks for the pics - that's interesting to see. I don't have any of module#1 style, but I'd hazard a guess the pot may be a different value, something like 50kΩ?

> > Good news is that you can try this simply by bridging the lower two pins of the R2 trimmer.
> 
> exactly and i did it before but unfortunately nothing change in the output of my pcb.

oh well, so much for easy solutions! 

This is actually a very forgiving chip and circuit, so I think it would be very unlikely that there's anything about your PCB construction causing the issue. I've even wired it on a breadboard with a DIP adapter for the MT3608.

At this point I'd actually be a bit more suspicious that the MT3608 chip is actually bad/damaged.
I think I'd go back to basics and either reproduce with fresh components on a breadboard/dead-bug or swap out the components on the PCB. I assume you probably have a few MT3608 chips on hand? If you can't get the circuit to work with any of them, I'd suggest getting a sample from another supplier.

Keep at it! I know the frustration when what should be a "simple" circuit won't behave, but the win is even more sweet;-)

behnam68, 2020-10-09:
@tardate  @jjohnson873 
Thanks a lot for helping and sharing your experience with me
at the end, i'll change my MT3608 and i'll order new one from other supplier and test it again. certainly i'll share the results here.

` I'd recommend you use flux next time since it doesn't appear you applied flux to the pc board before you soldered on the parts. You'll use less solder and the connections will be better.`
i consider your tips and in the next try i'll use solder mask on pcb to avoiding any unwanted solder short and also use flux.

`Also the inductor needs to be between 4.7uH to 22uH with low loss at 1.2MHz and a low DCR`
i'm using all exact component of a working module for my pcb but nothing happened.

`If you just had R2 fixed at 100K-ohm and R1 at 2.2K-ohm you should get a 28vdc output. Did you ever say what input voltage you're using?`
probably i'll design a fixed R2 pcb for next time because i just need the max output in the end. i'm using 5v input.

`At this point I'd actually be a bit more suspicious that the MT3608 chip is actually bad/damaged.
I think I'd go back to basics and either reproduce with fresh components on a breadboard/dead-bug or swap out the components on the PCB. I assume you probably have a few MT3608 chips on hand? If you can't get the circuit to work with any of them, I'd suggest getting a sample from another supplier.`
Sure, i'll do it. changing MT3608 and component... 

`Keep at it! I know the frustration when what should be a "simple" circuit won't behave, but the win is even more sweet;-)`
Exactly! i'm getting crazy because it must be incredibly simple and so easy, but takes me lots of money, time and energy...

Thanks again for taking the time to share your great information with me.

jjohnson873, 2020-10-09:
I think you'll be successful with your pc board.  It appears to be good, after making the one modification to the potentiometer connection.   You won't need to use a solder mask on the pcb.  Just use a small flux applicator or flux pen to apply flux to the pc board traces where your components are placed.  Your solder will flow much easier and smoothly.   Post again when you get your new components and try it out.

tardate, 2020-10-09:
Looking forward to hearing of your success @behnam68!

btw, I just threw the circuit back on a breadboard to make sure I wasn't mis-remembering - and yes, it works nicely, and as you can see in 
[Electronics101/SwitchModePowerSupplies/MT3608/Breadboarded](https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/SwitchModePowerSupplies/MT3608/Breadboarded), I didn't need to be too finicky about the wiring.

behnam68, 2020-10-12:
Hi dear friends
Finally new IC's arrived and i tested them with my designed PCBs and module's PCBs. they work great with variable configuration. I've tested with different capacitor values, fixed and variable output voltage with different pots and other component that i thing cause the problem.
there's difference between new and old ICs...according to the datasheet the top marking of ic must be : B628DC
[image]
and look at my ICs:
old and not working ic:
[image]
new one and working ICs:
[image]

new ICs are twice as expensive but i thing they're original.
another thing that i've tested was using flux. first soldering of new ic, the output voltage wasn't stable and not working as i expected. after some solder and remove component i realized the problem is my old liquid flux. the material left over from it after soldering isn't dielectric and pass the current between tracks especially in the small tracks of ic.So, i had to change it. with new flux and cleaning the PCB after soldering i'm getting a stable and expected output voltage. i repeated that process for other PCBs and components and they work great on the first try.

behnam68, 2020-10-12:
@tardate  i see you used the same new IC i bought
[image]

closed 2020-10-13 by behnam68

tardate, 2020-10-13:
@behnam68 excellent news, thanks for the update. The chips I'm using are labelled `B6286e`, which is probably similar to your `B6280T` as I think the last two chars may be a year/week code (according to one B628 datasheet I found). 

I got my stock on the cheap from aliexpress a while ago (5 for ~$2). I can't vouch for their provenance, but they do work.