tardate/LittleArduinoProjects #20: How configure the Keil for correct compile

mariliahoshino, 2021-01-07:
https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/PovShakeStickKit

I bought the similar circuit at Aliexpress
https://pt.aliexpress.com/item/33037796341.html
, and I'm using this for burn the program
https://www.aliexpress.com/item/1852124469.html?

But there are mistakes, how configure correctly the program Keil? I compiled in Keil and used PROGISP1.72 for burn the microcontroller 

https://github.com/mariliahoshino/POV/blob/main/Keil-compiler.png

I read the original program saved, and burn, it's OK, but when try burn a new program there are mistakes
https://github.com/mariliahoshino/POV/blob/main/error-burning.png?raw=true

thanks

mariliahoshino, 2021-01-16:
Finally I change the code and write my name

https://github.com/mariliahoshino/Shake_Stick_POV-PersistenceOfVision

closed 2021-01-16 by mariliahoshino

tardate, 2021-01-16:
@mariliahoshino good to hear you got it working, though I'm not a Keil user so couldn't help much (been using a DIY programmer and SDCC open source toolchain for 8051 programming).

Can you tell me which source code you used as a base for your modifications?

mariliahoshino, 2021-01-29:
I don't have the original code too
I modified direct at HEX code
https://github.com/mariliahoshino/Shake_Stick_POV-PersistenceOfVision
tks!!