tardate/LittleArduinoProjects #24: Improvements on the "Basic Discrete Schmitt Trigger"

steveschnepp, 2021-11-24:
I wonder how replicating the complexities of [7407 internals](https://www.ti.com/lit/ds/symlink/sn7407.pdf) would make it better. 

Asking since the schematics in the IC aren't **that** complex either, which rather uncommon.

tardate, 2021-11-26:
good idea @steveschnepp .. interesting to compare with the simple emitter-coupled design: aside from the input protection and open-collector output, I think this produces a more reasonable upper/lower switching window. I'll have to pull out Art of Electronics to see if they also went into that.

I'll put this in my todo list to investigate further and update my notes!

steveschnepp, 2022-02-21:
My bad. The 7407 is *not* a Schmitt Driver. The 747014 would be.

And I tried the schematic, it works nicely as a driver, but ita hystersis is very low.

closed 2022-02-21 by steveschnepp