tardate/LittleArduinoProjects #25: Improvements on the DicksonChargePump

steveschnepp, 2021-11-28:
I wondered if we could simplify the Dickson charge pump to only use 1 clock without the need to invert it.
And it seems it works indeed, as you can connect intermediate capacitors to ground instead of the inverted phase. 

It does charges much slower, but it works, and will save a pin on the Arduino. 

[image]

Access to an [interactive simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgbArCAMB00IQRhAFgJyzADgOwCZVUEx0l1tooIVroQIBTAWiSQCgATEdMEJXHwFDBIfPwByqJKmzsAxv1HClffBD70kLbCFaxyg7IQDM+E9DLrcMeNA7cVK9SLWTpshWI2rvfXjC2iPZiWMRUCDQC9gJB9lx+viYmrmLuMnKKyanZiVo6eiawEBDY0thlOFQmSCZxDiC5KtKp4khSGV4tvt0B9HDBKHD4uNgQJrjj5oJIFZp2HADuaEj+aj597ABKPOt8soko4hDBoVD4xYFwEOzLAS6+D9C3jz65T+wA5on3Pgf9212PV0Kn6aHoUDB13YACc0AFmgE+vxgs8AA4MIi+CBYhFXF4YVI41C+Z7LYlAwlA54mEkAL0YADtGDDWG1+LBcDITBB8NhyHykIh8CwbPQLqjJYgyvw2uw6WgwCSAqglY8QAzmaykPgCatKWrNstVcr9kjNASEaITaSCWrmvrQewAC4MMDi4jwtSerS2VC4Wm8MBmSDQFLzSb4MMtE74SA4oQIECcRgAMwAhgBXAA2zq+XsaUYLuQBxt+Zr4ZlLBcRakIMEB7t0hHoYAIYk9RxRZx7RRkSAg6Dj41K6Fw6Dq0LhTY7EPds8CQoQ7AA9vxqDrAsR0OgxP0DIv2EA).

tardate, 2021-11-29:
neat idea @steveschnepp - essentially a "half-wave rectifier" if I'm following properly. Nice compromise - twice the cycles to charge but save a pin.

steveschnepp, 2021-11-29:
I'm searching a way to have 0/5/12V from 1 pin. something like

| pin | rst |
| ---- | ------ |
| 0V | 0V |
| 5V | 5V | 
| 50kHz | 12V |

tardate, 2021-11-30:
@steveschnepp the closest I think I could get is using 2 pins (similar scheme for the 8051 programmer). PS: added a new project for the "half-wave" approach, see https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/Power/DicksonChargePump/HalfWave

closed 2021-11-30 by tardate

steveschnepp, 2021-11-30:
Don't stop a fool from doing impossible things.

Here is the circuit that does exactly this, with only 1 pin, guess my intuition was right.

Notes: 

- The input is a little bit changed, to be able to be able to emulate nicely "0V/5V/50kHz", but in real life, you would only use one GPIO.
- I cheated to keep the number of parts low, but it remains within the datasheet specs for the RESET line.

Access to an [interactive simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgbArOB0BMEFMC0AOEBmFtoHYzywEYAWdATjIAYxiQIw67K7lDCAoAExDIcLxD8+A2IIByxEinYBjQQKHyG8Bs0LI02SrEI1i9CLBTFKZWpWiVKHbosUqlIUYQlTZT+o4e8QajYOh0MAocUwgUa0JMc0trLg9hBnQgr3FJYmk5ZMSMFIc-VF9cQiJKHAowHBwIHH0iqxtcnMkc51cM9xbHLp9mWH8Lakh0HEIyErJwsnRYdHq4gHcQHuVPXvYAJWXNTz7PQicmKz7oKGwoZgsIdiX7T2yE33YAc0efBxpfJ62MxzUBS7LZgXL5XdgAJ3AJkcNGYvSOVnYAAc6MRaIoIGieHxQTdUeiBLC-niiT5MbR1uhaAAvBAAOwQ4KQhGcAVGpEMKDGRkIx36SBwXy0xxFosoxkEznY1LoLOxsucIhAtIZTJZeLRaDJcvWrwgcoe+tE6EMuIAzts3qtVIIQAAzACGABszQhNhgJbBoSbRF7AcxoSDLqcNbonN7PdDKBCFeHgXLFH1YIifpoUGg+unQUDGP6Q9wun7Y0W2ulpEsjfLyfLoxWEwJq4mYz5FL9E4JjuwLb8jFrlFm1Pbna68XdWgP4vYs7AcL7p2k3HIZ3PNLOml91IULCZk+QyCheDRooLBnFXsv1xefbiAC7bZgzePG5Mboo1cXjShBabDQgXaBgCMxBGLAvYzMBbZWCAnAII6ACuTo3nijYNli6CUOYMaFtCXToeYHadksGQPhhyxhnhTx3tEGAvrhL6DkgFizOSaKgdMJQmMBRQmGMKCYJQEBUAJxgpLyzAwfBiEvGRSQpLhpqXHi1GPh6tC9k8kIXup1HqWoIoahENEBoZFHRq8PgqRZClPLcIhZooKm1rGFHVqZzb9mgU4ZgR0aQg5L4WfRPlOWO8r5KGzRtgCEXypqNYalFDCVk2AD2gjoto+EAB5ctYoiDFUUzwPoJQzpMByntYr6iEg4zsEAA).

### RESET == LOW

At 0V, we have only 60 mV, but it should be enough to be seen 

[image]

### RESET == VCC

At 5V, we have only 4.3V, but it 

[image]

### RESET == HIGH VOLTAGE

[image]

tardate, 2021-12-01:
isn't that still 2 pins @steveschnepp ? (one for CLK, and one for switching +5V?)

steveschnepp, 2021-12-01:
No. 5V is VCC rail, which doesn't count like a pin.

And the 1K/10K resistor array is just for the ease of the simulation. In reality it is the CLK pin that will be 0, 5 or CLK.

tardate, 2021-12-02:
@steveschnepp ok got it - that's very clever;-)

I'll have to throw that on a breadboard!

steveschnepp, 2021-12-02:
My ultimate goal is to have a vusb atmega328 driving the programming. Thus not requiring a full blown Uno board.

My targets for programming are selfish :
- atmega328, with High Voltage erase
- at89cx051
- at24cXXX EEPROM

Ideally, leveraging avrdude (patched?) on the host side

steveschnepp, 2021-12-02:
Also, using a Schottky is only useful for the one that is between the 2K2 and the 10K resistors.
As the 12V doesn't need to be raised beyond, but it does raise the 4.3V to 4.5V. 

That said, I don't think it will make a huge difference, as the datasheet says "> 0.7 Vcc" is enough for a HIGH.

tardate, 2025-03-24:
Thanks @steveschnepp, I finally got around to testing the idea. Works great! Final circuit I used here: https://leap.tardate.com/electronics101/power/dicksonchargepump/onepinthreev/

steveschnepp, 2025-03-24:
I see that you managed to use MOSFETs. 
I didn't managed so I remained with BJTs, good job!