tardate/LittleArduinoProjects #28: #631 Long Running Timer Improvements

steveschnepp, 2022-05-04:
To have long delays, you are right, to avoid analogic error margins, we need to go digital. 

But I'm not sold on always using an MCU. It is the "too easy" road. 

As soon as you realize you will only leverage its counter/timer it makes a CD4060 rather well suited to the task. You can cascade them if _very_ long delays are needed, or use it with a quartz if precision is important. 

Only power of 2 are available, but if you go with a watch clock quartz of 32kHz, 2 CD4060 in cascade & some DIP-activated diode logic, you can have a very easy programmable timer. 

Note: if you really want to go the no-surprise road, you might need to leverage a CD4040 instead of the 2nd CD4060, as the latter has "holes" (to offer bigger divisions without increasing of the DIP form factor).

steveschnepp, 2022-05-04:
You even covered part of it in **#294 CD4060/CrystalOscillator** 😉

tardate, 2022-05-05:
good point @steveschnepp ! I've added your point to the discussion.

PS: I though the elektor article was good fodder for a simple little test, but I ended up banging my head for a few days trying to get the timer circuit to work with a 1GΩ resistor. Lesson learned;-)

closed 2022-05-05 by tardate

steveschnepp, 2022-05-06:
well, the NE555 that is used in the Elektoe mag (TLC556) has a leak discharge of only 0.1 nA max.

Whereas the standard one (LM555) has a leak on pin 7 of 100 nA max (with 1nA typical). Even the CMOS version has the same leakage characteristics.

So i guess you were fully right in that leakage was failing you. Having a aliexpress part didn't help, but wasn't even the main driver IMHO.

tardate, 2022-05-07:
thanks @steveschnepp - that's a really useful steer towards the leak discharge spec in the datasheets. I'll revisit the 1GΩ config if/when I get something like the TLC556 to play with

steveschnepp, 2022-05-07:
According to TI that IC was **specifically** designed for that 1Gohm scenario in mind.

> The TLC555 is a monolithic timing circuit fabricated using the TI LinCMOS™ process. The timer is fully compatible with CMOS, TTL, and MOS logic and operates at frequencies up to 2 MHz. Because of its high input impedance, this device supports smaller timing capacitors than those supported by the NE555 or LM555. As a result, more accurate time delays and oscillations are possible. Power consumption is low across the full range of power-supply voltage.
> from https://www.ti.com/product/TLC555#description

tardate, 2022-05-26:
@steveschnepp I tested the circuit with a TLC556C and the discharge leakage proved to be the critical factor. Works fine now, even on a breadboard. Updated the notes accordingly.