tardate/LittleArduinoProjects #30: [CD4069/LogicProbe] Another Improvement and some explanations

steveschnepp, 2022-05-24:
It's me again, sorry 😉 

I did not understand at first why you had to remove the input diode to make it work. That was not satisfactory. Therefore I had to look deeper into the circuit. I discovered quickly that the capacitor is not part of an _oscillator_, but is part of a _high-pass filter_. 
Once that is known, we can simply keep the high-pass filter and only buffer it with 2 NOT gates in input & output.

Note that, as we don't invert the filter anymore, we have to connect the filter resistor to the ground, and fully remove the feedback circuit. Note that here also, the input clamping diodes of the output NOT gate are helping the high-pass filter signal to stabilize. And if you really look into slow oscillation, you can see that the filter is actually an "monostable rise edge detector" 🤣.

The result is a **much cleaner** [version of the circuit](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l5AWAnC1b0DYQHZobEtgBwCsGJ2ATIZBSAliSCZMwKYC0YYAUAO7NsWSlRAZKCECMpQeYcWIngkM8ZOlSQAEzYAzAIYBXADYAXDsbZbwUWzEi95qpRrUgAzJVYydBk+ctrVghWe14AJ0VJAmdoyiJbMHgeSLdPVjSiSWDkgTcwePpPcELIHgBzKI8sqoQ4Ox4AJSKZAoSEYqJQ2zrmOyhoEib6JGFRBAQEjW96KUG5plDBngBjEC7vUSyp0WDYeAPDuDBObA9BhxICdxIWInd3MGxF-fgI9cnqyW2pXfAjuQKJK9ShKYGbGQ+PRGMwWKw2EIDBz8KptHBeEoJMoCKjBQo-NHY5gYbxkYkQ2QCMgU6lSIhYlG0jQUMYyIlXSG9Dm-NmMgo85j89yISmCyHjUYCsrNJlKWkxfpc-owIaVWnpckedxYUI8XRiqRckmG1gPcDzeyUPUG4WSdUi4pgC2wK367Dajwi4iSW0eVrOyCunAeiSsb0mrXml6Bio4Goa8MPHWyZrh0GScMK7pK7oqlHu1k4EXTfMh3oFqX5moaDaVnHV0Rp+mitNktO7a1ICaeyQoGS+x0BoNd9S9PsRs1O6NWyrjjVzj265rj9MgcdZnqsRZ2ZYCEcCoiSkt77safe+on70Nr08dvcqAVJSA7Xn38UyFfNy8Pyhkld34NCwrC8Un+Xo0XBKQlByOBY0PLANXgrVkzKawkMQo9RChPwzBRJCa0+Es0MwmR8JcbRoX8YYyO+SUN1YHMdyGAAjdYhA8EgPzAdxwHuWQAA91gwLB5CQf4HHAcYShAABJABhAAdABnGSADsAAdDFMZSABEAEsAHsdGUuTjH0ABbdS9NU8oUTcDRsGLACCUKSDCRRJ8XzAikykqH53BqH46lzHgDPACB5A8EpqC3JFjlsHisXWDxQvC8BkzADp3HdOKJPsSBPCIbjxEKjo4BuNlNG4FKAuSgAxCAeO8eouBAOrwjYABHQw2FUlYAE8eCAA), which I think is the minimal one now.

[image]

steveschnepp, 2022-05-24:
And, totally ironically, I just discovered that is even an already "well known circuit" : the [Leading Edge Detector](https://www.falstad.com/circuit/e-leadingedge.html). 🤦

tardate, 2022-05-27:
good stuff @steveschnepp  I'll try that...

tardate, 2022-09-19:
thanks @steveschnepp I finally got around to updating the project notes!

closed 2022-09-19 by tardate

steveschnepp, 2022-09-19:
Great job 👍

Btw, I'm wondering... You mentioned CD4069UBC, which is *Unbuffered*, but in the picture there is a CD4069BE, which should be *Buffered*.

So the various information out there is contradictory at best. Some saying exactly that difference and some saying that the CD4069 does *not* exists in a buffered version.

I'm wondering how to test if it is buffered or not 🤔

tardate, 2022-09-20:
good question.

It does seem the 4069 was only ever singly buffered / single-stage design (according the CMOS cookbook).

It does make one wonder what this apparently "TI CD4069BE" chip is I have in my parts box;-)
I can't find any reference to a "CD4069BE" from TI or in a bunch of old data books I found on archive.org.

I should put it under test for the variable output impedance one would expect with an unbuffered design.

steveschnepp, 2022-09-22:
Indeed. I am wondering about a test that would work to differentiate buffered from unbuffered.

steveschnepp, 2022-09-22:
Just asked on [EE.SE](https://electronics.stackexchange.com/questions/635840/how-to-test-if-an-ic-output-is-buffered)

steveschnepp, 2022-09-22:
And the answer is to simply do a X/Y transfer test. It would be interesting to compare:

* 1 CD4069UBE
* 3 CD4069UBE in series 
* 1 CD4001BE with its inputs tied