tardate/LittleArduinoProjects #35: Another Voltage Multiplier

steveschnepp, 2022-11-09:
Just discovered a Voltage Multiplier with a single CD40106/74HC14.

[[image]](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgDCDMB0Z-DEBYCmBaA7FaBWAnHpJAIwAcpkOAbAEwV5UhIQ4g4THrHEBQASiDqkQxPDUHURY8DOZsZEMLh4ATQeSnjK40eJ0A5JMSSkeAYxAZIjXZesikSBbAQE37jwXAvi3uLwBzOxtHYJE6BR4AdzDbUmIbaRoeGjAseMTxUhNNJhAAewBXABceACcQDNyqoRlieB4ErIT1YSrIVMEQFRQAMwBDQoAbYrQhlDVfRT9AypaOiHaccUVzKGXBOW1ciBpYUnQsJX8-MF4K7ZotjdqOOGg8cqYWq4gjRluRe8eLd823lq2Xb7Q6nKY+J4mXZyKGtOrfNaw15MUgcaTAsAHTBgsFPHChZH4pyfepKH5sAlyIk7QQg7HHM64ipUTH-EAs4QkhEWDls3lA2mY0EM8H+J5UDB6OQSvQaO5ktYyvmSml7IX0ny4tTbWx-WwGIwmVTPTIotF6ESGYymNSw2zU-WWw02ilOWz8pJO63Gj3iJWO4hWo1kSDrWXCP6fJSsYjQKiMGBEeRgRoUE1wpFy7wxuMJ6BJ1gpkNmuHUqO4ES57AF8Cp0NljS88s5+PV+u14tNxsq5uV1uJ9tFtP6jQ0SS92P9-ODtYj4S2IiMCAwRCrhCoNDCEW4oIL+wLjarGJz3JjpdPfXjq9LwSpFMxM8SD7juU8ADOT7hCzh0wGQzfKDRFAnS1Jcr7WMIOrSDqESrBB6a6oCsG1vBdrSHayEpvBDrSA6mE8PBvrsqyxD4Wo-rSFYpoGt6hEqu69H4fkeQdLIYB4MIhYuCQY4yKGTihpAW48MxkB4J+HA0DQeCiicDKMr4jByACsgrEwoZvIWBHCOJABiUwiOgW4uPAvhoL4ADC-QAA79GYACWxT9AAdmYgFCV8ID6XUXxwCA5kgHwKBvvZb5Oa57nznpBm+KSHAgAAOlh85YN58VxSIiXJSIrBpZlGW+El2kiKGeWxfAmVFR53BeTFvnxVVnIQGVRmnKZMiNSIwgtZubVyVlxUtD1xmIAZnXGLVdStfJ-XjeIw19YyEBFUAA)

Nothing out of the ordinary, it's just a regular "bucket brigade", but the Schmidt clock and the individual inverters makes it very scalable, as each stage has its own buffered output.

Note : I addded some resistors in the bucket capacitors to emulate current limitation from the IC.

tardate, 2025-02-25:
finally got around to trying this @steveschnepp Didn't have any CD40106 so used 74HC14 instead. Works like a charm, thanks! Published at [LEAP#593](https://leap.tardate.com/electronics101/74hc14/voltagemultiplier/)

steveschnepp, 2025-02-25:
Seems there is an ordering mess since the carving out of the model kits 😉

tardate, 2025-02-25:
> Seems there is an ordering mess since the carving out of the model kits 😉

I'm using smaller little projects like this to back fill the empty numbers:-)

steveschnepp, 2025-02-25:
Beware, I have a ton of other ideas..