tardate/LittleArduinoProjects #36: [#004 Pushbutton switch for LED] Use a NE555!

steveschnepp, 2022-11-13:
I actually had the exact same need. But didn't want to leverage a full MCU, a full Arduino even less 😉

So, I designed a schmitt trigger based one. Yet some search on the internet unearthed a NE555 based solution. Which is great as a DIP-8 is a smaller footprint as the DIP-14 from a CD40106. And using a NE555 for something is always nice 😂 

The trick is that the NE555 is also used as a schmitt trigger like my solution. I inverted the resistor values in my design, in order to consume less current (battery-powered). So a long press will actually keep toggling the output.

Here's the link to the NE555 one : https://www.electronics-lab.com/project/toggle-on-off-switch/

Mine is highly similar, but using a CD40106 and having 2x 1M resistors with a 47k one filling a 22uF capacitor. I used the other invertors in parallel to drive the LED with more current.

tardate, 2025-01-26:
thanks @steveschnepp , updated notes with ref to a 555 latch circuit (LEAP#692), but it seems I don't have an inverted-based latch circuit in the collection yet. I made a note to add one e.g. https://electronics.stackexchange.com/a/651628 - is that similar to your CD40106 circuit?

steveschnepp, 2025-01-28:
Here are the schematics of mine.

[image]

And the real thing

[image]

steveschnepp, 2025-01-28:
I used it as a reminder of "is a task done".

tardate, 2025-02-13:
Thanks for the pointers @steveschnepp . I made a similar circuit with a CD4069 in #117 and it works great.

closed 2025-02-13 by tardate

steveschnepp, 2025-02-14:
oh. the design is actually better than mine!

specially with the low pass / derive pass filters. 

thanks!