tardate/LittleArduinoProjects #39: Another, simpler, at89c2051 programmer

steveschnepp, 2023-04-18:
I got lazy and simply leveraged a 12v dc/dc converter. 
https://blog.pwkf.org/2022/08/27/simple-at89c2051-programmer.html

steveschnepp, 2024-11-24:
I just got a PCB I designed for it... 

[image]

Let's see if it works... i'll write another article about it

tardate, 2024-11-25:
that's a neat trick Steve! I've referenced it in the article, look forward to the follow-up post on the PCB

steveschnepp, 2024-11-25:
Note, the PCB use the [1-pin charge pump](https://github.com/tardate/LittleArduinoProjects/issues/25#issuecomment-982960430) I designed earlier. I agree with you that self contained is still better, and since it is a PCB, BOM simplicity is less important than with breadboarding.