tardate/LittleArduinoProjects #40: #464 Single Transistor AM Transmitter antenna length

harlock974, 2023-08-10:
Hello,
What is the length of the antenna used in the circuit ?

tardate, 2025-01-25:
@harlock974 many years later I answer your question (sorry).. the antenna is nothing special. I just used a ~20cm copper wire. Stefan0719 used a standard medium-wave antenna coil with ferrite core in his build. Neither of these are for efficient propagation - just enough to test with a nearby radio

closed 2025-01-25 by tardate

tardate, 2025-01-28:
> Hello [@tardate](https://github.com/tardate) , thank you for the answer. No worries for the delay, I have since studied the AM antenna question for my project : https://github.com/harlock974/time-signal

thanks, your notes on LF antennas are very educational reading