tardate/LittleArduinoProjects #42: [#736 CH340 on Apple Silicon] Driving a WS2812B

steveschnepp, 2024-03-27:
Now you can even drive a neopixel directly from your macos.. 🥳 

* https://blog.pwkf.org/2024/03/20/drive-ws2812-uart.html
* https://github.com/steveschnepp/ws2812bify

tardate, 2024-03-27:
that is very cool @steveschnepp .. I will have a go. how did I not find your blog already? the write-ups are excellent

tardate, 2026-11-12:
Thanks @steveschnepp, finally tried this myself! Found I couldn't use your scripts on macOS because it won't let you make a permanent baud rate change with `stty`. Out of interest, were you running your tests on Linux or Raspberry Pi?

closed 2026-11-12 by tardate

steveschnepp, 2026-11-12:
My laptop, on debian 13.

steveschnepp, 2026-11-12:
Also, I see that your usual 2N7000 is again better than my old tech NPN ones ;)

tardate, 2026-11-12:
> Also, I see that your usual 2N7000 is again better than my old tech NPN ones ;)

hehe. I actually tried the LED trick with an NPN but couldn't get it to work reliably. Maybe it came down to my resistor selection or other component differences. When I found that the nfet worked, I just went with the flow!

steveschnepp, 2026-11-12:
indeed. 2N7000 for the win again 🏆 

I do have some, but old habits of 2N3904 die hard... 

I really have to try the 2N7000 much more often as default.