tardate/LittleArduinoProjects #77: [#638 9V (PP3) to 5V or 3.3V Power Adapter] same with AA

steveschnepp, 2024-10-09:
I did mostly the same, but glued to a 2xAA holder for 3.3V and 4xAA holder for 5V.

This coupled with some [super small boost converters](https://aliexpress.com/item/1005006438496545.html), along with a beefy 1mF capacitor served most of my portable DC needs. 

Specially the 5V which even has directly a [female USB A connector](https://aliexpress.com/item/1189492730.html) so that i can do am emergency charge for. my phone with some still very easily sourced alcaline AA.

Bonus points is that if you don't need much current, you can put some dummy AA and reduce the number of real alcaline AA used.

tardate, 2024-10-10:
Thanks Steve. I've used the micro buck converters before but thanks for the link to their boost counter-parts - not seen these before, amazing. Have to get some and try out these ideas.

tardate, 2024-11-24:
I got some of those (tiny!) boost modules you mentioned. Very nice, tested: https://github.com/tardate/LittleArduinoProjects/tree/master/Electronics101/Power/AAA5vPower

closed 2024-11-24 by tardate