tardate/LittleArduinoProjects #84: [#642 USB 3 Power Adapter Board] Reverse!

steveschnepp, 2024-10-15:
Yeah, USB PD is really the way to go nowadays if you want more than 5V.

I'm using [those with a dipswitch now](https://www.aliexpress.com/item/1005004455694484.html). Works rather nicely, but it really depends on the quality of the usb pd charger.

I'm actually searching the reverse, some USB PD power provider than can take DC, for example from my usual stack of AA

tardate, 2024-10-15:
thanks for the link to the ones with DIP switches Steve;-)

> I'm actually searching the reverse, some USB PD power provider than can take DC, for example from my usual stack of AA

I guess closest would be one of the USB-C equipped power banks paired with one of these adapters, but I haven't seen any that go above 12V. If I come across anything like that I'll let you know

steveschnepp, 2024-10-15:
Note, if you want to test things with USB PD, you should have a look into 
* https://eeucalyptus.net/2023-12-06-usb-pd-1.html
* https://eeucalyptus.net/2024-05-13-usb-pd-2.html

tardate, 2025-01-25:
Thanks @steveschnepp, I also added the refs to the notes #106

closed 2025-01-25 by tardate